olikraus/U8glib_Arduino #1: Support for I2C on Teensy 3.1

fuse314, 2015-12-07:
I have a seeedstudio 128x64 OLED display that works with Adafruit's and seeedstudio's own library with Teensy 3.1
U8glib compiles for this combination, but I cannot get the display to show anything.
I used this initialization line:
U8GLIB_SSD1306_128X64 u8g(U8G_I2C_OPT_NONE|U8G_I2C_OPT_DEV_0);
I used the same wiring as with the other libraries (Hardware I2C pins)

Did anyone get this combination of devices to run? I would appreciate to get this very versatile library to work with my combination.

PS: I found the following dead thread in the pjrc forums (creator of the Teensy)
https://forum.pjrc.com/threads/23540-Using-u8glib-w-seeedstudio-I2c-oled-display-on-Teensy-3

clowrey, 2015-12-07:
Quite possibly this is because the u8glib library does not turn on the internal i2c pullups like the other libraries do. You can add external pullup resistors or alter the library to enable them.. I had this same problem.

clowrey, 2015-12-07:
Here is the issue I opened when it happened to me: https://github.com/olikraus/u8glib/issues/363

olikraus, 2015-12-07:
If i remember correctly, i never ported i2c to the Teensy.

derekr27, 2016-05-06:
I too have an OLED (SSD1306) that I'd like to use with the Teensy and U8G. Has i2c been ported to the Teensy yet, Oli?

Eheran1, 2017-01-10:
No update on this so far?
Cant get it to work on a Teensy 3.2.

olikraus, 2017-01-11:
That was one of the reasons, why i wrote u8g2. I2C support is available for Teensy with U8g2. Please migrate to U8g2.

closed 2018-04-29 by fuse314