olikraus/U8glib_Arduino #8: U8glib_Arduino fails to compile

csikiati, 2016-04-26:
Trying to compile Marlin firmware on Arduino 1.6.8, everything compiles well, but when it arrives to U8glib, this is what I got, indifferent of the directory the library is installed to: filenam or extencion too long.
```
Using library U8glib at version 1.19.1 in folder: C:\Program Files (x86)\Arduino\libraries\U8glib 
fork/exec C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc.exe: The filename or extension is too long.
Error compiling for board Arduino/Genuino Mega or Mega 2560.
```

olikraus, 2016-04-30:
I have no idea how to fix this.

csikiati, 2016-05-09:
Perhaps you shouldn't. I have got some feedback that this might be due to a Windows filesystem issue. (I have asked on stackoverflow hoping that there are people who have seen this themselves)
http://stackoverflow.com/questions/36844095/arduino-compile-error-of-u8glib-library-on-win-10/37029726#37029726
On Repetier it compiles, I can confirm, I am closing the issue.

closed 2016-05-09 by csikiati