olikraus/U8glib_Arduino #9: "The filename or extension is too long"

Treynolds416, 2016-07-28:
U8glib v1.19.1 has been compiling fine for me with Marlin while it's globally referenced through Sketch>Import Library in the Arduino IDE. However, if you try to reference it through a local directory you'll get the following error:

`fork/exec C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc.exe: The filename or extension is too long.
Error compiling for board Arduino/Genuino Mega or Mega 2560.`

I've also tried versions 1.17, 1.18 and 1.18.1 with the same result. Another person on the reprap forums seems to be having the same thing happen: http://forums.reprap.org/read.php?415,690652
It seems to be something specific to windows, but I'm not sure what the problem is past that.

olikraus, 2016-07-28:
The problem is not clear to me. In the Arduino IDE it is working, right?
According to your error message, this error is caused by avr-gcc.exe call. Is really a u8glib problem?

Treynolds416, 2016-08-08:
The problem is not very clear to me either. All I know is that if u8glib is referenced using `#include "u8glib.h"` (and relevant files are in the same directory) this error will appear, but if `#include <u8glib.h>` is used instead (and relevant files are in the right place) it will compile fine. I am able to reference other libraries locally with no issues, so I guessed that the problem was directly related to u8glib. I do apologize if the problem is not specific to it, I was just hoping for some insight because I don't really understand what would be causing this issue.
If you believe that it isn't a problem caused by u8glib, please close this issue.

olikraus, 2016-08-08:
strange... at least you have found a workaround.

closed 2016-08-08 by olikraus

bobc, 2017-09-21:
For reference, since I arrived via google.

## Analysis
This problem is due to Windows having a limit of 32767 characters for a command, and Arduino IDEs final step is a link with a _very long_ list of object files. If this exceeds 32k you get the fork/exec error.

u8lib is a red herring, it just happens to be a library with quite a few files.

## Workarounds
Altering the path to the source files causes the length of the link command to vary, so that is one way to get under the limit.
Another way to shorten the command is to change your %TEMP% location to somewhere as short as possible, e.g. "C:\TMP".

olikraus, 2017-09-21:
Thanks for the feedback

italocjs, 2018-08-20:
Just an update if someone is having issue with this,  update your arduino IDE to versio 1.9.0 beta at least, fixed those issues here.  (firmware v2.0.x)

outbound-pt, 2020-06-18:
> 
> 
> Just an update if someone is having issue with this, update your arduino IDE to versio 1.9.0 beta at least, fixed those issues here. (firmware v2.0.x)

Still won't work for me, and I really need to updated my machine's firmware because I changed the hardware configuration...

italocjs, 2020-06-18:
> > Just an update if someone is having issue with this, update your arduino IDE to versio 1.9.0 beta at least, fixed those issues here. (firmware v2.0.x)
> 
> Still won't work for me, and I really need to updated my machine's firmware because I changed the hardware configuration...

I ditched arduino ide quite some time ago, and it was totally worth it.  Try using the VSCODE + Platformio, i never had any weird bugs with it.

outbound-pt, 2020-06-19:
Thanks, I'm giving it a try right now.

dgiambo1, 2020-08-02:
Thanks, had the same problem VSCODE worked quite well. Used Auto Build Marlin Extension as well, worked first shot. Very nice tool.

Rudeboy2tone, 2020-08-08:
> For reference, since I arrived via google.
>
> ## Analysis
> This problem is due to Windows having a limit of 32767 characters for a command, and Arduino IDEs final step is a link with a _very long_ list [...]

I am having this issue and found this thread. It seems like you have a pretty good handle on the problem, but I don't really understand how to implement your solutions. 

Hoping you can help.

dgiambo1, 2020-08-08:
Rudeboy2tone: Follow this guide (https://marlinfw.org/docs/basics/install_platformio_vscode.html) to use VS code instead, it doesn't have the issues that Arduino IDE has.

dobromyslov, 2020-08-14:
I disabled REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER and the error does not bother me anymore.

nelson07br, 2020-09-09:
> I ditched arduino ide quite some time ago, and it was totally worth it. Try using the VSCODE + Platformio, i never had any weird bugs with it.

worked perfectly, thank you!

ashwi1987, 2021-05-04:
Hi All,
I have tried with the below link 
"https://docs.edgeimpulse.com/docs/running-your-impulse-arduino#code-compiling-fails-under-windows-os"
Go to "Code compiling fails under Windows OS" and follow the instruction 
I tried and reinstall the Arduino IDE and its working for me.
Thank You