olikraus/U8glib_Arduino #11: Cyrillic fonts for your library

sfrwmaker, 2016-08-20:
Dear sir,
I appreciate the library you created. I would like to add some new cyrillic fonts to your library. How can i send you a file with the font data?

olikraus, 2016-08-20:
Hi
The project is now continued here: https://github.com/olikraus/u8g2/

Font data should be a bdf file or a ttf file.
For a ttf file i need to know the desired target size.
In both cases, please provide the unicode ranges.

sfrwmaker, 2016-08-20:
Hello!

I have already converted the .bdf font into .c file and have put it into 
my copy of the library like this:

/* Cronyx WinCP1251 Fonts */
extern const u8g_fntpgm_uint8_t u8g_font_cronyx1c[] 
U8G_FONT_SECTION("u8g_font_cronyx1c");
extern const u8g_fntpgm_uint8_t u8g_font_cronyx1cb[] 
U8G_FONT_SECTION("u8g_font_cronyx1cb");
extern const u8g_fntpgm_uint8_t u8g_font_cronyx1cbo[] 
U8G_FONT_SECTION("u8g_font_cronyx1cbo");
extern const u8g_fntpgm_uint8_t u8g_font_cronyx1co[] 
U8G_FONT_SECTION("u8g_font_cronyx1co");
extern const u8g_fntpgm_uint8_t u8g_font_cronx1h[] 
U8G_FONT_SECTION("u8g_font_cronx1h");
extern const u8g_fntpgm_uint8_t u8g_font_cronyx1t[] 
U8G_FONT_SECTION("u8g_font_cronyx1t");
extern const u8g_fntpgm_uint8_t u8g_font_cronyx1tb[] 
U8G_FONT_SECTION("u8g_font_cronyx1tb");
extern const u8g_fntpgm_uint8_t u8g_font_cronyx1to[] 
U8G_FONT_SECTION("u8g_font_cronyx1to");
extern const u8g_fntpgm_uint8_t u8g_font_cronyx2c[] 
U8G_FONT_SECTION("u8g_font_cronyx2c");
extern const u8g_fntpgm_uint8_t u8g_font_cronyx2cb[] 
U8G_FONT_SECTION("u8g_font_cronyx2cb");
extern const u8g_fntpgm_uint8_t u8g_font_cronyx2cbo[] 
U8G_FONT_SECTION("u8g_font_cronyx2cbo");
extern const u8g_fntpgm_uint8_t u8g_font_cronyx2co[] 
U8G_FONT_SECTION("u8g_font_cronyx2co");
extern const u8g_fntpgm_uint8_t u8g_font_cronyx2h[] 
U8G_FONT_SECTION("u8g_font_cronyx2h");
extern const u8g_fntpgm_uint8_t u8g_font_cronyx2hb[] 
U8G_FONT_SECTION("u8g_font_cronyx2hb");
extern const u8g_fntpgm_uint8_t u8g_font_cronyx2hbo[] 
U8G_FONT_SECTION("u8g_font_cronyx2hbo");
extern const u8g_fntpgm_uint8_t u8g_font_cronyx2ho[] 
U8G_FONT_SECTION("u8g_font_cronyx2ho");
extern const u8g_fntpgm_uint8_t u8g_font_cronyx2t[] 
U8G_FONT_SECTION("u8g_font_cronyx2t");
extern const u8g_fntpgm_uint8_t u8g_font_cronyx2tb[] 
U8G_FONT_SECTION("u8g_font_cronyx2tb");
extern const u8g_fntpgm_uint8_t u8g_font_cronyx2tbo[] 
U8G_FONT_SECTION("u8g_font_cronyx2tbo");
extern const u8g_fntpgm_uint8_t u8g_font_cronyx2to[] 
U8G_FONT_SECTION("u8g_font_cronyx2to");
extern const u8g_fntpgm_uint8_t u8g_font_cronyx3c[] 
U8G_FONT_SECTION("u8g_font_cronyx3c");
extern const u8g_fntpgm_uint8_t u8g_font_cronyx3cb[] 
U8G_FONT_SECTION("u8g_font_cronyx3cb");
extern const u8g_fntpgm_uint8_t u8g_font_cronyx3cbo[] 
U8G_FONT_SECTION("u8g_font_cronyx3cbo");
extern const u8g_fntpgm_uint8_t u8g_font_cronyx3co[] 
U8G_FONT_SECTION("u8g_font_cronyx3co");
extern const u8g_fntpgm_uint8_t u8g_font_cronyx3h[] 
U8G_FONT_SECTION("u8g_font_cronyx3h");
extern const u8g_fntpgm_uint8_t u8g_font_cronyx3hb[] 
U8G_FONT_SECTION("u8g_font_cronyx3hb");
extern const u8g_fntpgm_uint8_t u8g_font_cronyx3hbo[] 
U8G_FONT_SECTION("u8g_font_cronyx3hbo");
extern const u8g_fntpgm_uint8_t u8g_font_cronyx3ho[] 
U8G_FONT_SECTION("u8g_font_cronyx3ho");
extern const u8g_fntpgm_uint8_t u8g_font_cronyx3t[] 
U8G_FONT_SECTION("u8g_font_cronyx3t");
extern const u8g_fntpgm_uint8_t u8g_font_cronyx3tb[] 
U8G_FONT_SECTION("u8g_font_cronyx3tb");
extern const u8g_fntpgm_uint8_t u8g_font_cronyx3tbo[] 
U8G_FONT_SECTION("u8g_font_cronyx3tbo");
extern const u8g_fntpgm_uint8_t u8g_font_cronyx3to[] 
U8G_FONT_SECTION("u8g_font_cronyx3to");
[... 160 lines cut ...]
   84,84,214,7,6,6,8,1,0,252,130,28,2,130,124,7,
   6,6,8,1,0,238,68,76,84,100,238,7,8,8,8,1,
   0,56,0,238,68,76,84,100,238,7,6,6,8,1,0,236,
   72,80,112,72,230,7,6,6,8,1,0,126,36,36,36,36,
   206,7,6,6,8,1,0,198,108,84,84,68,238,7,6,6,
   8,1,0,238,68,124,68,68,238,7,6,6,8,1,0,124,
   130,130,130,130,124,7,6,6,8,1,0,254,68,68,68,68,
   238,7,8,8,8,1,254,252,66,66,66,66,124,64,224,7,
   6,6,8,1,0,124,130,128,128,130,124,7,6,6,8,1,
   0,254,146,16,16,16,56,7,8,8,8,1,254,238,68,68,
   40,40,16,144,96,7,10,10,8,1,254,56,16,124,146,146,
   146,146,124,16,56,7,6,6,8,1,0,238,68,56,68,68,
   238,7,7,7,8,1,255,238,68,68,68,68,254,2,7,6,
   6,8,1,0,238,68,76,52,4,14,7,6,6,8,1,0,
   214,84,84,84,84,254,7,7,7,8,1,255,214,84,84,84,
   84,254,2,7,6,6,8,1,0,240,160,60,34,34,124,8,
   6,6,8,0,0,231,66,114,74,74,247,6,6,6,8,1,
   0,224,64,120,68,68,248,7,6,6,8,1,0,124,130,30,
   2,130,124,8,6,6,8,0,0,238,81,113,81,81,238,7,
   6,6,8,1,0,62,68,68,60,36,206};

Is it acceptable in the version 2 of your library?
Have you implemented the thick line frawing in the new version?

Best regards, Alexander.

olikraus, 2016-08-21:
If the font is available for the public, then i can integrate this to u8g2.
There is no think line drawing at the monent.

sfrwmaker, 2016-08-24:
Hello!

Here is the link: http://www.kovrik.com/sib/russify/x-windows/#win-fonts
Best regards. Alexander.

olikraus, 2016-08-27:
i have created an enhancement request in u8g2

sfrwmaker, 2016-08-28:
Thank you very much.