olikraus/U8glib_Arduino #12: Arduino 101 can use u8g library?

vyuke, 2016-10-24:
Hi，i want my oled drive by arduino 101,compilation is successful,connect is ok,but OLED doesn't work,i don't know why,please help me find the problem,thanks.
my code:
`
#include "U8glib.h"

#define REST 4
#define DC 5

U8GLIB_SSD1306_128X64 u8g(U8G_I2C_OPT_NONE);

void draw(void) {
// graphic commands to redraw the complete screen should be placed here

u8g.setFont(u8g_font_unifont);
//u8g.setFont(u8g_font_osb21);
u8g.drawStr( 20, 36, "OPENJUMPER");
}

void setup(void) {
pinMode(REST,OUTPUT);
pinMode(DC,OUTPUT);
digitalWrite(REST,HIGH);
digitalWrite(DC,LOW);

// flip screen, if required
// u8g.setRot180();

// set SPI backup if required
//u8g.setHardwareBackup(u8g_backup_avr_spi);

// assign default color value
if ( u8g.getMode() == U8G_MODE_R3G3B2 ) {
u8g.setColorIndex(255); // white
}
else if ( u8g.getMode() == U8G_MODE_GRAY2BIT ) {
u8g.setColorIndex(3); // max intensity
}
else if ( u8g.getMode() == U8G_MODE_BW ) {
u8g.setColorIndex(1); // pixel on
}
else if ( u8g.getMode() == U8G_MODE_HICOLOR ) {
u8g.setHiColorByRGB(255,255,255);
}
}

void loop(void) {
// picture loop
u8g.firstPage();

do {
draw();
} while( u8g.nextPage() );

// rebuild the picture after some delay
delay(500);
}`

olikraus, 2016-10-24:
101 will not work with U8glib. I have to use u8g2 (version 2 of u8glib)
https://github.com/olikraus/U8g2_Arduino

closed 2020-07-17 by vyuke

olikraus, 2020-07-17:
Nice, worth a comment after 4 years: Thanks for closing.