olikraus/U8glib_Arduino #15: slow performance in full buffer mode

Mulder8472, 2016-12-11:
hi,

either I don't understand the performace gain with full buffer mode or there is something wrong.

the Hello World exampe takes 583ms from clearBuffer to sendBuffer.

I am using this constructor:
U8G2_SSD1306_128X64_NONAME_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ SCL, /* data=*/ SDA, /* reset=*/ U8X8_PIN_NONE); 

I was hoping that I can use this to perform partial updates to the screen to increase performance. I.e. render a full screen and after that only update a small portion of the screen ( a rapidly changing progress bar). The old u8glib refreshed the entire screen in "just" 320 milliseconds (with HW fast I2C).

olikraus, 2016-12-11:
U8g2 would do the same speed in HW I2C mode:
U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0,/ reset=*/ U8X8_PIN_NONE);

Mulder8472, 2016-12-11:
nice, hw i2c brings it down to 60ms, is that running at 100 or 400khz?

only problem is that I run out of ram when using HW i2C :D

olikraus, 2016-12-12:
400khz if supported by your Board.
RAM: You can use the page version. 1 instead of F in the constructor.

Mulder8472, 2016-12-12:
I used u8glib in page mode before I switched to u8g2 and that took 330ms to render a screen in fast i2c mode.

The display will show frequency  and RSSI values (as bars) for a 4 channel lap timing system. it needs to render very fast, more than 10ms would too slow for me.

Yesterday i switched to u8x8 and am rendering the bar graphs with ascii (maybe tiles later) but it seems to be very fast, havent timed it yet, and uses way less ram, just not as pretty, but thats OK.

I am running an arduino nano, so it can barely run the hello world with full buffer :D 

sorry for writing all this in an issue ticket! is there a better place for asking this kind of questions?

olikraus, 2016-12-12:
> sorry for writing all this in an issue ticket! is there a better place for asking this kind of questions?

At least the u8g2 issue tracker would be better: 
https://github.com/olikraus/u8g2/issues

> more than 10ms would too slow for me

This is a frame rate of 100Hz. This is probably not achievable with any uC. In fact i guess the OLED refresh rate is only 70Hz.

Mulder8472, 2016-12-12:
i modified the u8x8 hello world example so the countdown starts from 1000 and removed the delay. it is roughly counting down 100 numbers per second, but could also be 70, seems pretty fast though. if the display cant handle it is not a problem, i just dont want the arduino busier for more than 10ms  at a time so i can continue timing the racers.

olikraus, 2016-12-12:
Great, indeed i guess u8x8 is the fasted method to output something.