olikraus/U8glib_Arduino #16: bdf2u8g_101 help please to use

VitaliGN, 2017-03-01:
Hello olikraus, 
I am sorry if I duplicate an issue, i did not find a solution to my problem. 
It is: I want to make my font, I finally found out how to make bdf file and now I am trying to use bdf2u8g_101.exe to make a file for u8g library. Is it a right way to do? I am trying to execute this exe but it loads and immediately  terminates. I have seen that there is in u8g2 library should be a new tool for this but there is no exe. Now I have 2 issues how to load an exe or how to use source files in font tools folder? How should I compile it? I am really sorry for being such a dumb. I need a help on converting bdf into u8g format to make it available for my screen.

olikraus, 2017-03-01:
Is this obsolete by your U8g2 question?

VitaliGN, 2017-03-01:
I will answer in some hours, I need to try, and may-be it will be easy to use existing fonts. but I think I will still have a question about how to convert fonts if it will be needed at all... However, may-be, if I will use u8g2 library then this issue should be there (excuse me).