olikraus/U8glib_Arduino #19: question re: delay instructions

kivikakk, 2017-12-23:
Hey there!

I'm using U8g2 on an ATmega328P and it's working great. I have a question re: the delay messages I'm accepting from U8g2. I first laid out my GPIO/delay callback with the following cases:

```c
    case U8X8_MSG_DELAY_NANO:
      _delay_us(0.001 * arg_int);
      break;

    case U8X8_MSG_DELAY_100NANO:
      _delay_us(0.1 * arg_int);
      break;

    case U8X8_MSG_DELAY_10MICRO:
      _delay_us(10 * arg_int);
      break;

    case U8X8_MSG_DELAY_MILLI:
      _delay_ms(arg_int);
      break;
```

I noticed that e.g. doing a `sendBuffer` with full framebuffer worked fine, but updated slowly. For some reason I decided to try this:

```c
    case U8X8_MSG_DELAY_NANO:
      //_delay_us(0.001 * arg_int);
      break;

    case U8X8_MSG_DELAY_100NANO:
      //_delay_us(0.1 * arg_int);
      break;

    case U8X8_MSG_DELAY_10MICRO:
      //_delay_us(10 * arg_int);
      break;

    case U8X8_MSG_DELAY_MILLI:
      //_delay_ms(arg_int);
      break;
```

And it still works, but faster!

Do you have any idea why? One suggestion that has been made to me is that, because I'm running on the internal clock @8MHz, it's not running fast enough that the "extra" delays are needed. I plan to get the AVR running from an external clock @16MHz soon, and maybe that'll break without the delays.

Thanks for your time!

olikraus, 2017-12-23:
Yes, AVR systems are slow enough and delay might not be needed at all.

kivikakk, 2017-12-23:
Lovely! Thank you so much for your quick response, and your amazing library. :heart:

closed 2017-12-23 by kivikakk