olikraus/U8glib_Arduino #20: ST7735

gekzakm, 2018-02-08:
It is worth waiting for the support of the display ST7735?
Is it possible to correct an existing library to support this display?

olikraus, 2018-02-08:
I moved out RGB display support to a different library: https://github.com/olikraus/Ucglib_Arduino

gekzakm, 2018-02-13:
tell me what function?
firstPage();
nextPage();
I downloaded the library, but it doesn't have these features. how can I adapt?

olikraus, 2018-02-13:
I do not support u8glib any more. What about using Ucglib as mentioned above?

gekzakm, 2018-02-13:
I have a program code that contains these functions
but there is used this library, but there is no support for my screen
help to understand and convert the code under the library for my display

olikraus, 2018-02-14:
This is a open source library. I do not earn any money here. And (as mentioned before) I do not support this library any more. My time on answering any support requests is limited. Moreover I am not able to guess what you want.

So again, here are my answers, hopefully clear enough:

> It is worth waiting for the support of the display ST7735?

No, ST7735 will not be implemented with u8glib

>  Is it possible to correct an existing library to support this display?

Yes, this is a open source library, but I will not do this.

gekzakm, 2018-02-14:
I downloaded the library by reference.
Start the display.
Perhaps you do not understand the question.
There are no functions in the library by reference
firstPage();
nextPage();

is there an analog of these functions in the library that you suggested to use?

olikraus, 2018-02-14:
> is there an analog of these functions in the library that you suggested to use?

No