olikraus/U8glib_Arduino #22: avr-gcc.exe: The filename or extension is too long

praveenmuru, 2019-02-06:
fork/exec C:\Users\tom4p\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\5.4.0-atmel3.6.1-arduino2/bin/avr-gcc.exe: The filename or extension is too long.
Error compiling for board Arduino/Genuino Mega or Mega 2560.

I have seen previous comments like changing the %temp% to C: \temp. But please help me out with where to update this and how?

Polar12345, 2020-10-06:
I am having the same issue trying to compile for the creality/cr-10s
fork/exec C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc.exe: The filename or extension is too long.
Error compiling for board Arduino Mega or Mega 2560.

windumasta, 2021-02-08:
Hi Same here trying to compiel for un reprap with adruino mega and ramp1.4