olikraus/U8glib_Arduino #24: Error variable 'colors' with dynamic initialization put into program memory area

djmitho, 2020-06-21:
On the example U8GLib.ino is not possible to compile. 
Error on line 142: const colorDef<uint8_t> colors[] MEMMODE={

full error of compiler is:

_src/menuIO/U8GLibOut.h: In function 'global constructors keyed to 65535_0_U8GLib.ino.cpp.o':
U8GLib:142:25: error: variable 'colors' with dynamic initialization put into program memory area
 const colorDef<uint8_t> colors[] MEMMODE={
                         ^
lto-wrapper.exe: fatal error: C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc returned 1 exit status
compilation terminated.
c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld.exe: error: lto-wrapper failed
collect2.exe: error: ld returned 1 exit status
exit status 1
variable 'colors' with dynamic initialization put into program memory area_

Arduino version 1.8.13
Any idea?

olikraus, 2020-06-22:
I think this is the wrong place to ask. U8gLibOut.h is not a file which was created inside this project.
Moreover U8glib is replaced by U8g2.

djmitho, 2020-06-24:
I am sorry about that
I will try today to change that in the program
Thank you

djmitho, 2020-06-24:
Hello again
With U8G2 library i get the same error compilation
Do you know how to solve this problem?
Is with the example included on library "ArduinoMenu" called U8G2

C:\....\Arduino\libraries\ArduinoMenu-master\src/menuIO/u8g2Out.h: In
function 'global constructors keyed to 65535_0_U8G2.ino.cpp.o':
U8G2:80:25: error: variable 'colors' with dynamic initialization put into
program memory area
 const colorDef<uint8_t> colors[] MEMMODE={
                         ^
lto-wrapper.exe: fatal error: C:\Program Files
(x86)\Arduino\hardware\tools\avr/bin/avr-gcc returned 1 exit status
compilation terminated.
c:/program files
(x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld.exe:
error: lto-wrapper failed
collect2.exe: error: ld returned 1 exit status
exit status 1
variable 'colors' with dynamic initialization put into program memory area

Thank you in advance

olikraus, 2020-06-26:
Not sure what I can do here. I did not wrote the code. Better to ask the author of u8g2Out.h.

djmitho, 2020-06-28:
ok thanks

haklein, 2021-08-25:
you can't use `colors[]` in later versions but would need something like this:
`const colorDef<uint16_t> colors[6] MEMMODE = {
  {{(uint16_t)BLACK, (uint16_t)WHITE}, {(uint16_t)BLACK, (uint16_t)WHITE, (uint16_t)WHITE}}, //bgColor
  {{(uint16_t)WHITE, (uint16_t)BLACK}, {(uint16_t)WHITE, (uint16_t)BLACK, (uint16_t)BLACK}}, //fgColor
  {{(uint16_t)WHITE, (uint16_t)BLACK}, {(uint16_t)WHITE, (uint16_t)BLACK, (uint16_t)BLACK}}, //valColor
  {{(uint16_t)WHITE, (uint16_t)BLACK}, {(uint16_t)WHITE, (uint16_t)BLACK, (uint16_t)BLACK}}, //unitColor
  {{(uint16_t)WHITE, (uint16_t)BLACK}, {(uint16_t)BLACK, (uint16_t)BLACK, (uint16_t)BLACK}}, //cursorColor
  {{(uint16_t)BLACK, (uint16_t)WHITE}, {(uint16_t)WHITE, (uint16_t)BLACK, (uint16_t)BLACK}}, //titleColor
};`