olikraus/U8glib_Arduino #27: SAMD: delay.h: No such file or directory

red-scorp, 2021-01-01:
Following happens when build for M0 or Zero boards both in Arduino SDK and platform-io.

```
C:\Users\***\Documents\Arduino\libraries\U8glib\src\clib\u8g_delay.c:57:14: fatal error: delay.h: No such file or directory
#    include <delay.h>
              ^~~~~~~~~
compilation terminated.

```
It helps to comment the include out:
```
/* issue 353 */
//#if defined(ARDUINO_ARCH_SAMD)
//#    include <delay.h>
//#endif
```

Could you fix it yourself or should I go a full circle with cloning and issuing pull request?

olikraus, 2021-01-02:
I think this is more kind of a platform-io problem.

red-scorp, 2021-01-02:
The issue is, that Arduino SDK also reports the error.
Judging from the form, the include was inserted in the file, it was a quick fix for some compile problem which was occurring with SAMD SDK. Which means, now its the time to remove it.
So, how we will proceed with this? Compilation of u8g is now on SAMD platform is impossible.

olikraus, 2021-01-02:
hmm.. actually I do not support u8glib any more. Did you check u8g2?

red-scorp, 2021-01-02:
Will you release new version if I fix the stuff?

olikraus, 2021-01-04:
Probably not. Many many things got fixed with u8g2, so I assume it is better to use u8g2 instead.