olikraus/U8glib_Arduino #28: AVR GCC fails to compile the library

zezic, 2021-03-03:
When trying to build the project using this lib with arduino-makefile I got this:
```
/usr/share/arduino/hardware/tools/avr/bin/avr-g++ -MMD -c -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=105  -D__PROG_TYPES_COMPAT__ -I/usr/share/arduino/hardware/arduino//cores/arduino -I/home/zezic/Work/Toolchanger/toolchanger/Arduino_Boards/sparkfun/avr/variants/promicro    -I/home/zezic/Work/Toolchanger/toolchanger/libs-arduino/SPI/src    -I/home/zezic/Work/Toolchanger/toolchanger/libs-arduino/Wire/src    -I/home/zezic/Work/Toolchanger/toolchanger/libs/U8glib_Arduino/src -Wall -ffunction-sections -fdata-sections -Os -DUSB_VID=0x1b4f -DUSB_PID=0x9206 -fpermissive -fno-exceptions -std=gnu++11 -fno-threadsafe-statics -flto main.cpp -o build-promicro-16MHzatmega32U4/main.cpp.o
In file included from /home/zezic/Work/Toolchanger/toolchanger/libs/U8glib_Arduino/src/U8glib.h:42:0,
                 from main.cpp:4:
/home/zezic/Work/Toolchanger/toolchanger/libs/U8glib_Arduino/src/clib/u8g.h:123:25: internal compiler error: in type_hash_canon, at tree.c:6953
 typedef uint8_t PROGMEM u8g_pgm_uint8_t;
                         ^

/home/zezic/Work/Toolchanger/toolchanger/libs/U8glib_Arduino/src/clib/u8g.h:123:25: internal compiler error: Segmentation fault
avr-g++: internal compiler error: Segmentation fault (program cc1plus)
```
It compiles if I do the next change, but I've not checked the functionality yet:
```diff
-typedef uint8_t PROGMEM u8g_pgm_uint8_t;
+typedef uint8_t u8g_pgm_uint8_t;
```

olikraus, 2021-03-07:
U8glib is not supported any more. Did you try u8g2?

zezic, 2021-03-07:
Thank you for pointing me! Would be nice to have a notice about new version to inform other repository visitors.
It seems like u8g2 works fine for me except the only one thing – I don't have `yield` function implemented in my Arduino core library. Probably, I'll have to add a header with empty implementation in it to be able to compile the lib without modifications.