olikraus/U8glib_Arduino #30: Anet A8 BigTreeTech Issue

ulises269, 2022-04-23:
I am trying to get the BigTreeTech mini12864 working on the Anet A8. No matter what I try the same error pops up. 
it says U8glib-HAL.h does not exsist 
I'm using the Marlin Firmware 2.0.9.3

olikraus, 2022-04-23:
U8glib is an Arduino lib, the mentioned h file is not part of this project

ulises269, 2022-04-23:
> U8glib is an Arduino lib, the mentioned h file is not part of this project

I have the U8glib in my Arduino Lib. But can't find that h file anywhere

olikraus, 2022-04-23:
> But can't find that h file anywhere

Correct, because it is not required for Arduino Boards.

ulises269, 2022-04-23:
> > But can't find that h file anywhere
> 
> Correct, because it is not required for Arduino Boards.

I'm confused. So then can I not run the BigTree LCD or am I just doing something incorrectly?

olikraus, 2022-04-23:
Marlin has created a special port of U8glib, which not supported by this original project. U8glib-HAL.h is an extension created by Marlin project. This is the original project, which is not at all related to Marlin and especially can not help with any questions related to U8glib-HAL.h

ulises269, 2022-04-23:
> Marlin has created a special port of U8glib, which not supported by this original project. U8glib-HAL.h is an extension created by Marlin project. This is the original project, which is not at all related to Marlin and especially can not help with any questions related to U8glib-HAL.h

Thank you so much for the help. I would've gone crazy trying to make this work