olikraus/U8glib_Arduino #31: U8glib.h: No such file or directory

MrIceWizard, 2023-04-15:
Hello, I'm using Arduino IDE and I'm uploading to a MKS 1.4 board (Arduino Mega basically) and the IDE is saying U8glib.h: No such file or directory. I'm not sure what I'm doing wrong, could someone help?

olikraus, 2023-04-15:
I don't support u8glib any more. What about using u8g2?

MrIceWizard, 2023-04-15:
Would I just replace U8glib.h with U8g2 in the code?

olikraus, 2023-04-15:
If you have an existing code, then replacing U8glib.h with U8g2.h would require some porting.
Alternative solution: If U8glib,h is missing, then you need to install u8glib in the Arduino IDE.

MaxyMeanderings, 2023-07-07:
> Hello, I'm using Arduino IDE and I'm uploading to a MKS 1.4 board (Arduino Mega basically) and the IDE is saying U8glib.h: No such file or directory. I'm not sure what I'm doing wrong, could someone help?

Hey, I'm not sure if you ever fixed this, but I had the same error. I fixed it by Installing the library directly from the Arduino IDE. 
The "U8GLIB-HAL" by Scott Library seems to be the correct one.